bistromath/gr-air-modes #68: modes_rx SBS-1 crashes on Linux mint x64

engink1981, 2015-05-08:
hi Nick,

I have tried  latest committed version, but it gives another new error after 20 minutes. I'm using  1090 MHz roof antenna and receiving lots of ads-b transponder.( receiving diameter is around 700 km)

Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/air_modes/zmq_socket.py", line 101, in run
    self._pubsub[address] = msg
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/pubsub.py", line 52, in __setitem__
    sub(val)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/parse.py", line 433, in publish
    pub["type%i_dl" % ret.data.get_type()] = ret
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/pubsub.py", line 52, in __setitem__
    sub(val)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/sbs1.py", line 95, in output
    sbs1_msg = self.parse(msg)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/sbs1.py", line 142, in parse
    outmsg = self.pp4(msg.data, msg.ecc)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/sbs1.py", line 165, in pp4
    aircraft_id = self.get_aircraft_id(ecc)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/sbs1.py", line 86, in get_aircraft_id
    for icao, _id in map(self._aircraft_id_map).iteritems():
TypeError: map() requires at least two args

bistromath, 2015-05-08:
yeah i'm the worst. sorry, i can't test over here as i'm off on vacation,
so i'm pushing untested code. can you try it again?

--n

engink1981, 2015-05-08:
I have tried three times, result is same.

bistromath, 2015-05-08:
After pulling the latest and reinstalling again?

engink1981, 2015-05-08:
sure, I have uninstalled previous installation and recompiled again.

engink1981, 2015-06-15:
hi  bistromath ,

did you try latest sbs1 python code?

closed 2016-08-15 by bistromath