bistromath/gr-air-modes #71: modes_rx --help crashes on OS X 10.10 with Fatal Python Error

nguillermin, 2015-06-13:
> $ modes_rx --help
> Fatal Python error: PyThreadState_Get: no current thread
> Abort trap: 6

Worthwhile to note that OS X's own error reporter then came on. I'll spare you the gigantic error message I got from that (also, I forgot to copy it).

nguillermin, 2015-07-21:
I believe but can't confirm that the problem ended up being the librtlsdr version, problem which was documented elsewhere.

Solution is to `brew install librtlsdr --HEAD`.

closed 2015-07-21 by nguillermin

bistromath, 2015-07-21:
Thank you for the update.

nguillermin, 2015-07-26:
Solution up there didn't actually fix anything. After taking a look at the makefile output I tried this:

    cmake -D PYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib \
    -D PYTHON_EXECUTABLE=$(which python) \
    ../

My suspicions were that it had been compiled across different libraries so I thought I'd manually specify them. Following this I was able to `modes_rx --help` successfully.