bistromath/gr-air-modes #73: Can't decode by filename source

shianglin, 2015-06-27:
hi Nick, 
I name is Siang Lin, I have one problem during I run the modes_rx by filename source.
the problem shows below:
command : modes_rx  --source="/home/cnsl/raw1.dat"
output:
Using file source /home/cnsl/raw1.dat
Using Volk machine: avx_64_mmx_orc
Segmentation fault

what' wrong with "Segmentation fault"?
Thx!
Best Regards~

bistromath, 2015-06-30:
What format is the data you're feeding gr-air-modes? It is expecting
complex float32 format. If you feed it data formatted in a different way
the Gnuradio file source can segfault.

shianglin, 2015-06-30:
Hi Nick, 
Thank you for your responds.

I try the two different ways to record the data. One is I recorded data by  usrp_source and file sink in gnuradio-companion application, and the usrp_source output format is complex float-32. The other way is I use "rx_sample_to_file" to record the data in "float" (fc32) format.  In additionally, I also utilized the older gr-air-modes in 3.6 version gnuradio to replay this data.  The data can be replayed by this version.  So, I make sure that  the data is complex-float32.

Thx!
Best Regards~

closed 2015-07-01 by shianglin

bistromath, 2015-07-01:
Did you figure this out?

shianglin, 2015-07-01:
Yes, the problem is in the pmt::pmt_t tag_to_timestamp of preamble_impl.cc.
If the source is from file, the code line in "if(tstamp.key == NULL || pmt::symbol_to_string(tstamp.key) != "rx_time") " will return 0. Thus, the "tstamp.value" can not be obtained. Therefore, I add the few code in function of "static pmt::pmt_t tag_to_timestamp(gr::tag_t tstamp, uint64_t abs_sample_cnt, int rate)" as shown below:
if(tstamp.key == NULL || pmt::symbol_to_string(tstamp.key) != "rx_time"){
last_whole_stamp = 0;
last_frac_stamp = 0;
{
else{
....
{

bistromath, 2015-07-01:
Can you submit a pull request?

Thanks!

shianglin, 2015-07-01:
Sorry, what is the mean?