bistromath/gr-air-modes #74: pyuic4 not found, not installing GUI application

bram-s, 2015-07-13:
Hi, I can't install the gui version of gr-air-modes on OSX 10.10.4:

anonymous:build anonymous$ cmake -DPYTHON_LIBRARY=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DCMAKE_INSTALL_PREFIX=/opt/local ../
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   thread
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_RUNTIME_FOUND = TRUE
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.6.
-- Found PythonLibs: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib (found version "2.7.6") 
-- Found PythonLibs: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib (found suitable version "2.7.6", minimum required is "2") 
-- pyuic4 not found, not installing GUI application
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/anonymous/Documents/SDR/gr-air-modes/build

But pyqt4 is installed and the output of cmake/Modules/FindPyQt.py is:
anonymous:Modules anonymous$ python FindPyQt.py 
pyqt_version:040b04
pyqt_version_str:4.11.4
pyqt_version_tag:Qt_4_8_6
pyqt_sip_dir:/opt/local/share/py27-sip/PyQt4
pyqt_sip_flags:-x VendorID -t WS_MACX -x PyQt_NoPrintRangeBug -t Qt_4_8_6 -x Py_v3 -g

bistromath, 2015-07-15:
I don't know about Mac, but on Ubuntu at least pyuic4 is not bundled with
pyqt4 -- you have to also install pyqt4-dev-tools. There's probably a
similar package for Mac.

--n

devnulling, 2015-08-21:
@bram-s 
If you used macports to install, check if pyuic4 is located in:
`/opt/local/Library/Frameworks/Python.framework/Versions/2.7/bin`

If so, add this line to your ~/.profile file and then rebuild.
`export PATH="/opt/local/Library/Frameworks/Python.framework/Versions/2.7/bin:$PATH"`

closed 2016-08-15 by bistromath