bistromath/gr-air-modes #81: Keep getting this error on 'make'

dsound, 2016-04-29:
<[  3%] Building CXX object lib/CMakeFiles/air_modes.dir/preamble_impl.cc.o
[  7%] Building CXX object lib/CMakeFiles/air_modes.dir/slicer_impl.cc.o
[ 11%] Building CXX object lib/CMakeFiles/air_modes.dir/modes_crc.cc.o
[ 15%] Linking CXX shared library libair_modes.so
[ 15%] Built target air_modes
make[2]: *** No rule to make target 'swig/air_modes_swig.py', needed by 'swig/air_modes_swig.pyc'.  Stop.
CMakeFiles/Makefile2:196: recipe for target 'swig/CMakeFiles/pygen_swig_f461b.dir/all' failed
make[1]: *** [swig/CMakeFiles/pygen_swig_f461b.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2>

darkstar007, 2016-04-29:
I also had this error. I found a work around - rename the gr-air-modes directory to be gr-air-modes-0.0
not exactly why - must be lost in the depth of the cmake stuff - I tried to look for it but rapidly lost the will to live!

I'd be interested to hear if that work-around works for you.

dsound, 2016-04-29:
I tried your suggestion but now the Cmake files says it doesn't recognize gr-air-modes-0.0

darkstar007, 2016-04-30:
OK, so I did:

- git clone https://github.com/bistromath/gr-air-modes.git
- mv gr-air-modes gr-air-modes-0.0
- cd gr-air-modes-0.0
- mkdir build
- cd build
- cmake ..

bistromath, 2016-04-30:
What platforms are you both running on?

dsound, 2016-04-30:
Kali Linux Rolling All. I've installed about every package needed for installing SDR and Gnuradio stuff. I looked for the missing Swig package but nothing showed up in Synaptic.

darkstar007, 2016-04-30:
Debian 'unstable'.