bistromath/gr-air-modes #83: Map interface no longer works

devnulling, 2016-05-15:
The Google Map display appears to no longer work.

Getting this error in the console:

file:///tmp/mode_s.html line 120: ReferenceError: Can't find variable: MarkerWithLabel

Google Maps appears to require an API key now, and this script 404's when it's trying to be loaded - http://google-maps-utility-library-v3.googlecode.com/svn/tags/markerwithlabel/1.1.9/src/markerwithlabel.js

devnulling, 2016-05-17:
I've submitted PR #84 that fixes this issue.

bistromath, 2016-05-17:
A pull request! Thanks!

closed 2016-05-17 by bistromath

sbmueller, 2016-06-19:
The map interface does not work for me as well. Error is:
`http://maps.google.com/maps/api/js?key=None line 32: Google Maps API error: InvalidKeyMapError https://developers.google.com/maps/documentation/javascript/error-messages#invalid-key-map-error`

devnulling, 2016-06-20:
@sbmueller You need to register for an API key with Google Maps API @ https://developers.google.com/maps/documentation/javascript/ and put it into the API key field.