bistromath/gr-air-modes #94: too many junk message

hawkingyy, 2016-11-01:
I am using RTL2832U on Kubuntu16.04.
When I am running gr-air-modes with gain 25, 2M/s sample rates, I could decode so many message without position, callsign. In the begining, I thought may be I should take more time, however, as time goes on, I could find few useful message. I have tried different gain, sample rates, but resulted the same. 
`MSG,6,0,44,583092,144,2016/11/01,12:42:45.777,2016/11/01,12:42:45.777,,,,,,,,1340,0,0,0,0
MSG,7,0,45,E46C5E,145,2016/11/01,12:42:45.980,2016/11/01,12:42:45.980,,21300,,,,,,,,,,0
MSG,7,0,46,82E5F0,146,2016/11/01,12:42:47.294,2016/11/01,12:42:47.294,,95700,,,,,,,,,,0
MSG,7,0,47,30D269,147,2016/11/01,12:42:47.901,2016/11/01,12:42:47.901,,61400,,,,,,,,,,0
MSG,6,0,48,B02692,148,2016/11/01,12:42:48.206,2016/11/01,12:42:48.206,,,,,,,,0014,0,0,0,0
MSG,5,0,49,0631D0,149,2016/11/01,12:42:51.437,2016/11/01,12:42:51.437,,14800,,,,,,,1,0,0,1
MSG,6,0,50,1CB7A9,150,2016/11/01,12:42:51.841,2016/11/01,12:42:51.841,,,,,,,,2032,0,0,1,
MSG,6,0,51,BC8AAE,151,2016/11/01,12:42:51.842,2016/11/01,12:42:51.842,,,,,,,,4056,0,0,0,0
MSG,7,0,52,2CEB2C,152,2016/11/01,12:42:52.348,2016/11/01,12:42:52.348,,15100,,,,,,,,,,1
MSG,7,0,53,CDDC29,153,2016/11/01,12:42:52.450,2016/11/01,12:42:52.450,,-1300,,,,,,,,,,0
MSG,5,0,54,B571A4,154,2016/11/01,12:42:52.853,2016/11/01,12:42:52.853,,31200,,,,,,,0,0,0,1
MSG,5,0,55,66BF2D,155,2016/11/01,12:42:53.258,2016/11/01,12:42:53.258,,29300,,,,,,,0,0,0,1
MSG,6,0,56,0D4003,156,2016/11/01,12:42:53.360,2016/11/01,12:42:53.360,,,,,,,,0300,0,0,0,0
MSG,5,0,57,DCDC33,157,2016/11/01,12:42:54.371,2016/11/01,12:42:54.371,,111100,,,,,,,1,0,1,
MSG,7,0,59,4FC65E,159,2016/11/01,12:42:56.190,2016/11/01,12:42:56.190,,-1300,,,,,,,,,,0
MSG,5,0,60,5CD3DF,160,2016/11/01,12:42:56.291,2016/11/01,12:42:56.291,,36600,,,,,,,0,0,0,0
MSG,6,0,62,8AE599,162,2016/11/01,12:43:00.129,2016/11/01,12:43:00.129,,,,,,,,0260,0,0,0,1
MSG,6,0,63,7027A3,163,2016/11/01,12:43:00.836,2016/11/01,12:43:00.836,,,,,,,,1002,1,0,0,0
MSG,5,0,64,E94F71,164,2016/11/01,12:43:00.836,2016/11/01,12:43:00.836,,4700,,,,,,,0,0,0,0
MSG,7,0,65,90608E,165,2016/11/01,12:43:01.047,2016/11/01,12:43:01.047,,15000,,,,,,,,,,1
MSG,5,0,68,5969B7,168,2016/11/01,12:43:03.273,2016/11/01,12:43:03.273,,8900,,,,,,,1,0,1,
MSG,7,0,69,6FDDF8,169,2016/11/01,12:43:04.487,2016/11/01,12:43:04.487,,40975,,,,,,,,,,1
MSG,7,0,70,195E4C,170,2016/11/01,12:43:04.487,2016/11/01,12:43:04.487,,57900,,,,,,,,,,1
MSG,7,0,71,DF14F1,171,2016/11/01,12:43:05.393,2016/11/01,12:43:05.393,,4700,,,,,,,,,,0
MSG,7,0,72,A200A2,172,2016/11/01,12:43:05.999,2016/11/01,12:43:05.999,,47800,,,,,,,,,,1`

Using the same RTL2832U in windows with rtl1090.exe, everything goes alright. I don't know why?
`MSG,7,111,11111,7807BD,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,24175,,,,,,,,,0
ID,,111,11111,7807BD,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,CSC8295
MSG,1,111,11111,7807BD,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,CSC8295,,,,,,,,,,,
MSG,4,111,11111,780C73,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,,440.0,358.0,,,0,,,,,
MSG,7,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,34100,,,,,,,,,0
MSG,8,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,,,,,,,,,,,0
MSG,7,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,34100,,,,,,,,,0
MSG,3,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,34100,,,30.57546324,104.4208190,,,,,,0
ID,,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,CES2206
MSG,1,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,CES2206,,,,,,,,,,,
MSG,4,111,11111,780D95,111111,2016/11/01,12:44:57.000,2016/11/01,12:44:57.000,,,461.0,5.0,,,0,,,,,
MSG,7,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,24275,,,,,,,,,0
MSG,5,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,24250,,,,,,,,,,0
MSG,8,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,,,,,,,,,,,0
MSG,7,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,24225,,,,,,,,,0
MSG,3,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,24275,,,31.43914691,104.4244445,,,,,,0
MSG,4,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,,411.0,358.0,,,1536,,,,,
MSG,5,111,11111,7807BD,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,24250,,,,,,,,,,0
MSG,5,111,11111,780DFA,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,10975,,,,,,,,,,0
MSG,4,111,11111,780DFA,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,,342.0,266.0,,,-576,,,,,
MSG,5,111,11111,780DFA,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,10975,,,,,,,,,,0
MSG,7,111,11111,780C73,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,,,,,,,0
MSG,5,111,11111,780C73,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,,,,,,,,0
MSG,8,111,11111,780C73,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,,,,,,,,,,,0
MSG,7,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,,,,,,,0
MSG,5,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,,,,,,,,0
MSG,8,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,,,,,,,,,,,0
MSG,3,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,30.57760464,104.4210388,,,,,,0
MSG,4,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,,461.0,5.0,,,0,,,,,
MSG,5,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,,,,,,,,0
MSG,5,111,11111,780D95,111111,2016/11/01,12:44:59.000,2016/11/01,12:44:59.000,,34100,,,,,,,,,,0`
Besides, the select box "Show ADS-B-equipped aircraft" only can not be selected.

hawkingyy, 2016-11-01:
you can even see negative height in messages decoded by gr-air-modes

bistromath, 2016-11-01:
gr-air-modes uses a relatively sensitive threshold to pick out preambles.
Use the -T parameters to set a higher threshold.

hawkingyy, 2016-11-01:
@bistromath
I have tried to set threshold to 10, then I got fewer junk messages, but still can not find useful message contain position and callsign.
In windows, the threshold of rtl1090.exe is about -75dB. In real ADS-B receiver, the threshold is about -90dB. I don't know why we should take 10dB in gr-air-modes.
Besides, when I take off the antenna，and use default threshold, I can still receive lots of messages, but that can not happen when I use rtl1090.exe in windows.

bistromath, 2016-11-01:
The threshold is defined as the minimum preamble energy over the noise
floor.

hawkingyy, 2016-11-01:
@bistromath
In these so many messages, I can not find the same  HexIdent(Aircraft Mode S hexadecimal code, Field 5). That means no flight report ADS-B message twice or more, but that can not happen.

hawkingyy, 2016-11-01:
where is wrong?

hawkingyy, 2016-11-01:
without antenna, it can still receive so many message, I think there is something wrong with the drive of RTL2832U.

coolcow, 2016-11-14:
I confirm the same problem with my hackrf one !

Some negative altitudes, no coordinates, disabled "Show ADS-B-equipped aircraft" checkbox.
I have a strong signal and receive 200/300 messages/s, but it seems that something is going wrong while decoding.

I can reproduce the problem on a pentoo livecd (64bit).

bistromath, 2016-11-14:
Try using the DC block option. HackRF has a pretty huge DC offset.

coolcow, 2016-11-14:
DC Block option doesn't fix the problem.

bistromath, 2016-11-14:
Ok. What you're receiving are spurious replies that happen to pass CRC.
It's not a problem in decoding, it's a noise problem. Try increasing the
decoder threshold until you don't see anything when the antenna is
disconnected. Then back off the threshold a little until you very maybe 5
or 6 false replies per minute. Then try it with the antenna.

If that doesn't do it you might have strong local or internal interference
at or near 1090MHz. Take a look at the received spectrum with osmocom_fft.
1090MHz, 8Msps.

hawkingyy, 2016-11-15:
@bistromath,
When I am using modes_gui, from Live Data, I can also see junk message, but the program goes alright at visible aircraft. How do you filter useful message from large numbers of junk message?

closed 2017-03-27 by bistromath