bistromath/gr-air-modes #97: Xubuntu 16.04 error

sincos364, 2016-11-17:
sten@sten:~$ sudo modes_rx
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

Traceback (most recent call last):
  File "/usr/local/bin/modes_rx", line 99, in <module>
    main()
  File "/usr/local/bin/modes_rx", line 58, in main
    tb = air_modes.modes_radio(options, context)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/radio.py", line 47, in __init__
    self._setup_source(options)
  File "/usr/local/lib/python2.7/dist-packages/air_modes/radio.py", line 167, in _setup_source
    self._u = uhd.single_usrp_source(options.args, uhd.io_type_t.COMPLEX_FLOAT32, 1)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/__init__.py", line 122, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 1981, in make
    return _uhd_swig.usrp_source_make(*args)
RuntimeError: LookupError: KeyError: No devices found for ----->
Empty Device Address

bistromath, 2016-11-17:
This just means it's not finding whatever radio you're using it with.

closed 2016-11-17 by bistromath

sincos364, 2016-11-17:
ok