bistromath/gr-air-modes #98: QtWebKit no longer in PyQt4 (for example, in Ubuntu 16.10)

ac2ie, 2016-11-23:
tom@kudu:~$ modes_gui
Traceback (most recent call last):
  File "/usr/local/bin/modes_gui", line 24, in <module>
    from PyQt4 import QtCore,QtGui,QtWebKit
ImportError: cannot import name QtWebKit

Work-arounds on the web suggest you can get it to work by manually installing an earlier version of the PyQt4 package (I didn't try this), or use QtWebKit from PySide (I tried this but it didn't work):

tom@kudu:~$ modes_gui
Traceback (most recent call last):
  File "/usr/local/bin/modes_gui", line 33, in <module>
    from air_modes.modes_rx_ui import Ui_MainWindow
  File "/usr/local/lib/python2.7/dist-packages/air_modes/modes_rx_ui.py", line 428, in <module>
    from QtWebKit.qwebview import QWebView
ImportError: No module named QtWebKit.qwebview

Links to the Debian 8 discussion seems to say developers need to switch to PyQt5.

Krastanov, 2016-12-09:
Commenting out a few lines inside of the file permitted me to launch the software without crashes. The map does not work, as it is the part that requires the WebView.

I will see whether I can spend some time on making it work with PyQt5.

@ac2ie , could you please share the link to the Debian 8 discussion you mentioned?

ac2ie, 2016-12-10:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=784513

User TheCompiler suggest QtWebKit will eventually be removed from PyQt5: http://askubuntu.com/questions/840318/importerror-no-module-named-pyqt4-qtwebkit

niehauss, 2017-03-27:
Krastanov, since I am unexperienced with pyhon but the bug hurts me too: can you suggest which lines to commemt out?

Thanks, 

Sebastian

bistromath, 2017-03-27:
Not a lot I can do about this other than reimplement the map outside QtWebKit. I'll make the map support conditional on having QtWebKit installed, so it at least runs when the map is unavailable.

Krastanov, 2017-03-27:
There are some options that do not require reimplementation, just renaming some of the imports and fixing some API calls. I am sorry for the vagueness, I will try to respond more completely briefly.

JWCampbell, 2017-07-25:
Any update on this issue? If nobody else is working on it, I will probably take a try at it, but it would be better if I knew I wasn't duplicating another persons effort.

bistromath, 2017-07-25:
I was holding off on this until Gnuradio finally moves to Python 3. Also
because I'm lazy. =) If you'd like to take a stab at it I'm happy to merge
it in.

jayeye, 2019-05-26:
no progress on this bug in two years? The problem persists. Does nobody use qt-air-modes?

bistromath, 2019-05-28:
The modes_gui component was never much used, and I haven't had the
bandwidth to maintain it to keep up with GUI toolkit changes. I should
probably deprecate it.