bistromath/gr-air-modes #106: Please switch to Python 3

rhertzog, 2019-09-20:
gnuradio is now working with Python 3 and Debian is actively dropping support for Python 2. This means that gr-air-modes (which is packaged in Debian) also needs to use Python 3 instead of Python 2.

See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=936650 for the Debian bug report.

It would be nice to have a new tagged release working with Python 3. If that new release could also fix #98 that would be nice, even if you decide to deprecate modes_gui it's OK.

rhertzog, 2019-09-20:
BTW, for Qt GUI in Python 3 you have the choice between PyQt5 and Pyside2.

bistromath, 2019-09-20:
I just pushed a version which builds and runs on Gnuradio 3.8 (which is Python 3-only).

It's untested as I'm on the road right now. I'll tag it and close this after testing. Reports of success or failure would be welcome!

rhertzog, 2019-10-17:
Did you have the time to test your update? Unfortunately, we are only "packagers" and not real users so we can't really help with the testing.

bistromath, 2019-10-17:
Master modes_rx now works with Python3 and GR3.8. Tag "gr37" works with
Python2 and GR3.7. I'm working on updating modes_gui to transition to
PyQt5, remove Qwt and replace with custom widgets, and reimplement the map
display. This will take some time.

Nick

closed 2020-06-24 by bistromath