bistromath/gr-air-modes #108: modes_rx crashes on ubuntu 20.04 & python3.8.2

lweiheng, 2020-05-02:
after about 5 mins , it shows:

Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/zmq_socket.py", line 104, in run
    self._pubsub[address] = msg
  File "/home/jacob/sdr/lib/python3/dist-packages/gnuradio/gr/pubsub.py", line 55, in __setitem__
    sub(val)
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/parse.py", line 432, in publish
    pub["type%i_dl" % ret.data.get_type()] = ret
  File "/home/jacob/sdr/lib/python3/dist-packages/gnuradio/gr/pubsub.py", line 55, in __setitem__
    sub(val)
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/sql.py", line 77, in insert
    query = self.make_insert_query(message)
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/sql.py", line 93, in make_insert_query
    query = self.sql17(msg.data)
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/sql.py", line 122, in sql17
    [altitude, decoded_lat, decoded_lon, rnge, bearing] = air_modes.parseBDS05(data, self._cpr)
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/parse.py", line 285, in parseBDS05
    [decoded_lat, decoded_lon, rnge, bearing] = cprdec.decode(data["aa"], data["lat"], data["lon"], data["cpr"], 0)
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/cpr.py", line 221, in decode
    self.weed_poslists()
  File "/home/jacob/sdr/lib/python3/dist-packages/air_modes/cpr.py", line 196, in weed_poslists
    for key, item in poslist.items():
RuntimeError: dictionary changed size during iteration

jaredd, 2020-05-05:
I'm getting a similar error in under a minute on containerized Debian buster.  Installed gr-air-modes from git in a docker container.  Fed with rtl_tcp.  Messages seem to decode but then this error.

```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3/dist-packages/air_modes/zmq_socket.py", line 104, in run
    self._pubsub[address] = msg
  File "/usr/lib/python3/dist-packages/gnuradio/gr/pubsub.py", line 55, in __setitem__
   sub(val)
  File "/usr/local/lib/python3/dist-packages/air_modes/parse.py", line 432, in publish
   pub["type%i_dl" % ret.data.get_type()] = ret
  File "/usr/lib/python3/dist-packages/gnuradio/gr/pubsub.py", line 55, in __setitem__
    sub(val)
  File "/usr/local/lib/python3/dist-packages/air_modes/msprint.py", line 147, in handle17
    [altitude, decoded_lat, decoded_lon, rnge, bearing] = air_modes.parseBDS05(msg.data, self._cpr)
  File "/usr/local/lib/python3/dist-packages/air_modes/parse.py", line 285, in parseBDS05
    [decoded_lat, decoded_lon, rnge, bearing] = cprdec.decode(data["aa"], data["lat"], data["lon"], data["cpr"], 0)
  File "/usr/local/lib/python3/dist-packages/air_modes/cpr.py", line 221, in decode
    self.weed_poslists()
  File "/usr/local/lib/python3/dist-packages/air_modes/cpr.py", line 196, in weed_poslists
    for key, item in poslist.items():
RuntimeError: dictionary changed size during iteration
```
```bash
$ uname -a 
Linux 6ef5ea8f9423 5.4.0-7626-generic #30~1588169883~20.04~bbe668a-Ubuntu SMP Wed Apr 29 21:00:02 UTC  x86_64 GNU/Linux
```

bistromath, 2020-05-05:
Looks like a bug introduced by the conversion to Python 3. I'm modifying the CPR position list in place to prune old position reports, but it seems like Python 3 doesn't like that.

closed 2020-06-24 by bistromath