bistromath/gr-air-modes #110: PyQt4 not found, not installing GUI application

BismaAmjad, 2020-07-15:
I got this error during cmake:

**CMake Error at CMakeLists.txt:91 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.8".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.9.0.0-git**

So I changed the version in CmakeLists.txt to  3.9.0.0, and ran cmake again. The above error was gone however a new error appeared:
**-- Build type not specified: defaulting to release.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Using install prefix: /usr/local
-- Building for version: 9e2515a5 / 1.0.0git
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen missing components:  dot
CMake Warning at res/CMakeLists.txt:28 (find_package):
  By not providing "FindPyQt4.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PyQt4", but
  CMake did not find one.

  Could not find a package configuration file provided by "PyQt4" with any of
  the following names:

    PyQt4Config.cmake
    pyqt4-config.cmake

  Add the installation prefix of "PyQt4" to CMAKE_PREFIX_PATH or set
  "PyQt4_DIR" to a directory containing one of the above files.  If "PyQt4"
  provides a separate development package or SDK, be sure it has been
  installed.

-- PyQt4 not found, not installing GUI application
-- Configuring done
-- Generating done
-- Build files have been written to: /home/itu/gr-air-modes/build**

I'm using Ubuntu 18.04 and the same error appears for GNU Radio 3.7.13.4.

bistromath, 2020-07-17:
Looks like it's working OK. The GUI application isn't supported right now anyway -- does it build if you type "make"?

BismaAmjad, 2020-07-20:
The above issue was produced in a remote system that I've been using and for some unknown reason I'm unable to replicate the exact cmake error in that system as the cmake fails to configure this time. 
However, this is the output error I'm getting on my personal system (GNU version = 3.7.14.0, UH= 3.14.0), and it shows incomplete configuration error for cmake:

dat@dat-OptiPlex-9020:~/gr-air-modes/build$ cmake ../
-- Build type not specified: defaulting to release.
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so;/usr/lib/x86_64-linux-gnu/libvolk.so
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FFT
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_PMT_FOUND = TRUE
-- Extracting version information from git describe...
CMake Error at /usr/local/lib/cmake/gnuradio/GrMiscUtils.cmake:146 (install):
 **install TARGETS given no LIBRARY DESTINATION for shared library target**
  "air_modes".
Call Stack (most recent call first):
  lib/CMakeLists.txt:49 (GR_LIBRARY_FOO)

-- Using install prefix: /usr/local
-- Building for version: ... / ...
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.9", minimum required is "2") 
CMake Warning at res/CMakeLists.txt:28 (find_package):
  By not providing "FindPyQt4.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PyQt4", but
  CMake did not find one.

  Could not find a package configuration file provided by "PyQt4" with any of
  the following names:

    PyQt4Config.cmake
    pyqt4-config.cmake

  Add the installation prefix of "PyQt4" to CMAKE_PREFIX_PATH or set
  "PyQt4_DIR" to a directory containing one of the above files.  If "PyQt4"
  provides a separate development package or SDK, be sure it has been
  installed.

-- PyQt4 not found, not installing GUI application
-- Configuring incomplete, errors occurred!
See also "/home/dat/gr-air-modes/build/CMakeFiles/CMakeOutput.log".
dat@dat-OptiPlex-9020:~/gr-air-modes/build$ make
make: *** No targets specified and no makefile found.  Stop.

**Since you mentioned that GUI isn't supported anyway so I ignored the PyQt4 error, and from [here](https://stackoverflow.com/questions/14990343/cmake-error-targets-given-no-library-destination-for-shared-library-target) I used third answer to resolve this error in the cmake 'install TARGETS given no LIBRARY DESTINATION for shared library target' . When i ran cmake again, this is what i get: **
[... 142 lines cut ...]
  swig/CMakeLists.txt:39 (GR_SWIG_MAKE)

CMake Error at /usr/local/lib/cmake/gnuradio/GrSwig.cmake:178 (include_directories):
  Error evaluating generator expression:

    $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>

  Target "gnuradio::runtime_swig" not found.
Call Stack (most recent call first):
  swig/CMakeLists.txt:39 (GR_SWIG_MAKE)

-- Generating done
-- Build files have been written to: /home/dat/gr-air-modes/build
dat@dat-OptiPlex-9020:~/gr-air-modes/build$ make
[  4%] Linking CXX shared library libair_modes-....so
/usr/bin/ld: cannot find -lgnuradio::gnuradio-runtime
collect2: error: ld returned 1 exit status
lib/CMakeFiles/air_modes.dir/build.make:146: recipe for target 'lib/libair_modes-....so.1.' failed
make[2]: *** [lib/libair_modes-....so.1.] Error 1
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/air_modes.dir/all' failed
make[1]: *** [lib/CMakeFiles/air_modes.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

**Does this seem like the  compatibility issue with GNU Radio version?, Does this ADSB module works only for GNU 3.8?**

bistromath, 2020-07-20:
Currently only 3.8 is supported. As 3.9 matures I'll add support for it in a separate branch.

BismaAmjad, 2020-07-21:
Thanks, Installing GNU radio 3.8 solved the issue. But the command is not found when I run _modes rx_ , although everything is installed without errors. 

dat@dat-OptiPlex-9020:~/gr-air-modes/install/build/bin$ modes_rx -r 10e6
Command 'modes_rx' not found, but can be installed with:
sudo apt install gr-air-modes

bistromath, 2020-07-22:
Look to see where it was installed. Run "sudo make install" and check to
see what path it used.

BismaAmjad, 2020-07-24:
Yup. Did exactly like that and fixed it. Thanks a lot.

closed 2020-07-24 by BismaAmjad

Zaibi9999999, 2024-04-08:
I have been experiencing the same issue regarding the PyQt4 error (ADS-B) for the past four weeks. Although I installed GNU Radio version 3.8.1.0, it gives me the same error. Can you help me to resolve this error?